cucapra/filament #26: Enforcing stronger bounds on dynamic events

rachitnigam, 2022-08-10:
The following program type checks:
```
import "primitives/core.fil";

component main<G, H, L>(
  @interface[G, G+3] go_G: 1,
  @interface[H, L] go_H: 1,
  @interface[L, L+1] go_L: 1,
  @[H, L] in: 32
) -> () where G >= H, L >= G+3 {
  M := new Mult;
  m0 := M<G>(in, in);
}
```

The constraints on `H` and `L` ensure that the requirements of `m0` are met. However, these constraints are not enough to make the program pipeline safe.

This is because `G` may re-trigger after `3` cycles but `L` from the first iteration may still be active. This means that the second iteration of this pipeline would not be able to satisfy the requirement `H <= G`. 

I think we need to say something about the constraints on *the constraints themselves*. Specifically, given a bunch of constraints, we need to make sure that they are correct for *this execution* as well as the pipelined execution. In this case, I think the thing to enforce is that `G + dG >= H + dH` and `L + dL >= G + 3 + dG`, where `dX` is the delay of the interface `X`. These constraints ensure that the requirements implied by the first set of constraints are satisfiable during pipelined execution.

sampsyo, 2022-08-15:
Attempting to follow along with the timeline of events here, I drew this crappy ASCII art diagram that tries to number the cycles and put the events on the cycles they might occur:

```
0123456789
H  G  LLLL
   ^--m0
^------in
```

I used `LLLL` to mean the `L` event could be anywhere in there. Not sure this is intelligible at all, but it helped me a bit.

Maybe this is naive, but is it a problem that a second invocation in this circumstance could not satisfy the constraints? That is, a client would be prevented from making a second invocation of `main` where the second invocation's `G` occurs before the previous iteration's `L` is finished. Or maybe this is not possible to check?

rachitnigam, 2022-08-19:
I think you got the problem setup right! When thinking about type safety, its useful to distinguish between two things:
1. The safety properties of the explicit invocations, i.e., conflicts b/w same instances in the same pipeline *iteration*.
2. The safety properties of *implicit invocations*, i.e. conflicts b/w an instance b/w *different* pipeline iterations.

I think you are referring to checking (1) which I think is doable. However, (2) cannot be explicitly checked this way. It needs to be implicitly encoded using constraints on `@interface` of the events itself.

In this case, I *think* we need to say something like:

$$
G >= H \implies G + \delta G >= H + \delta H
$$

$$
L >= G + 3 \implies L + \delta L >= G + \delta G + 3
$$

This is essentially putting some bounds on the delays of various events and forcing them to be related so that the corresponding `@interface` for them ensure that cross iteration usage of pipeline components will not conflict.

Does that make any sense?

sampsyo, 2022-08-20:
Yes, roughly speaking! It does make me wonder whether those "implicit invocation" should be opt-in, in the sense that not every module needs to account for them, and some explicit marker would instruct the type-checker to account for pipelining.

rachitnigam, 2022-08-20:
Hm, I'm not sure if opt-in behavior would fit with the rest of the language. The implicit pipelining comes from the fact that we use events (like `G` and `L`) to schedule program execution and specify a minimum delay before they retrigger. We can always choose a bigger than required delay for the events but even in that case, we must assume that the events retrigger at that point creating the implicit pipelining.

Removing this behavior would somehow mean that an event can never retrigger under this paradigm.

sampsyo, 2022-08-22:
OK, good point. I guess I'm just trying to think about whether there's a way to naturally extend the checking of individual invocations for overlap to defensively allowing infinite re-invocation… but maybe this is not a productive avenue.

rachitnigam, 2022-08-23:
Yeah, this is something I've been wondering about for a while. I think there are two challenges to this:
1. Checking cross-iteration conflicts is a forall property: we need to ensure that no future iteration conflicts with the current one
2. If there are two invocations A and B of the same instance and $A_i$ and $B_i$ refer to their use in the $i^{th}$ iteration, then we need to check conflicts between $A_i$ for all $i$ as well as $\forall i \forall j A_i \bot B_j$ as well. In this way, reusing resources creates weird new constraints that are not quite the same as intra-iteration conflicts

rachitnigam, 2022-08-24:
One possible lead is invocations: maybe the delay of formal arguments and actual arguments should induce some relationship between the delays of the corresponding events. In the above example, the formal port `m0.left` has a delay of `3` while the actual port `in` has a delay of `|L-H|`. Maybe there should be some constraint generated about how these are related to ensure that `L` ends before the next `H` shows up

rachitnigam, 2022-08-25:
Thinking about this more, the problem is that constraints are only checked for *intra-iteration* parts of the pipeline (i.e., its body) but not *inter-interation* parts. In other words, constraints need to influence the delays of events that they mention.

For example, the following program is incorrect:
```
import "primitives/core.fil";

component main<G, L>(
  @interface[G, G+3] go_G: 1,
  @interface[L, L+3] go_L: 1,
  @[G, G+1] in_G: 32,
  @[L, L+1] in_L: 32,
) -> () where L >= G+3 {
  M := new Mult;
  mg := M<G>(in_G, in_G);
  ml := M<L>(in_L, in_L);
}

// This program needs to be have the following history:
// G ---> G + 3 ---> L ---> L + 3 ----> G'
```
This is because of the following accepted history:
1. g0 at cycle 0
2. l0 at cycle 3, g1 at cycle 3
The history obeys all constraints but is wrong because it creates a conflict.

Instead, the interface for `G` needs to be influenced by the constraint `L >= G+3`

rachitnigam, 2022-08-26:
This is probably not possible to do. See https://github.com/cucapra/filament/issues/27

closed 2022-09-02 by rachitnigam