cucapra/filament #69: Using indicator variable for constraint checking

rachitnigam, 2023-02-14:
## Using Indicator Variables

Currently, we use the following loop to check whether a given constraint `P` is valid, i.e., satisfied by all assignments to the variables:
```
(push)
(assert (not P))
(check-sat)
(pop)
```

Instead of doing that, we can attempt to generate "indicator variables" which essentially tell us which constraints can be violated.
For example, we can have a constant `t0` along with the formula:
```
(assert (and t0 (not P)))
```

This formula is satisfiable if and only if both `t0` and `P` can be assigned to true at the same time. This means that if we allocate a different variable for each assert and the `check-sat` query says `SAT`, we can look at the model and look which indicator variables were set to `1` to know which constraints were violated.

## Getting all Violated Constraints

Of course, in the above model, we will only get a non-deterministic subset of indicator variables set to `1` when the constraints are violated. In order to get all the violated constraints, we can ask the solver to maximize the value of `(sum t0 .. tn)` which would force it to assign as many `ti` to 1 as possible.

rachitnigam, 2023-02-14:
Okay, the above encoding doesn't actually work. You need something like:
```
(assert (or (not P1) (not P2) ...))
(assert (= (= t0 1) (not P1)))
(assert (= (= t1 1) (not P2)))
...
```

The first assert asks whether any assert `P1 ... Pn` is a contradiction while the second forces the indicator variables to be set to `1` when there is a contradiction.

This alone is still not enough to get all the contradictions; we need to do a binary search over the sum of `t1 ... tn` to find all violated constraints.

rachitnigam, 2023-02-15:
This doesn't work unfortunately. Let's say that there are `n` contradictions in the original set. To get the solver to find this set, we'd hope we can just ask it to find a model where the sum of the indicator variables is `n`. However, what we're actually telling it to do is find *one* assignment which falsifies all `n` statements at the same time which might not be possible :-(

closed 2023-02-15 by rachitnigam

dz333, 2023-02-15:
I'm quite sure that Andrew would try to pitch SHERRLOC to you if he heard you had this problem, since that's exactly what it's for (i.e., assigning blame when a program fails to satisfy a constraint system) and I know Siqiu recently spent a lot of time making SHERRLOC more usable, but I don't know where the code is or how to use it :joy:

rachitnigam, 2023-02-15:
Right! I just want "reasonable" error messages but that's gotten pretty annoying even in the absence of type inference. I think it's definitely going to get way worse once I add inference support to Filament *sigh*.  CC @andrewcmyers in case this discussion seems interesting

rachitnigam, 2023-02-15:
#70 attempted to implement this but it didn't work

andrewcmyers, 2023-02-15:
If you can encode your analysis into SHErrLoc's constraint system then you could just let it do the checking and the error diagnosis.  Otherwise you might be able to adapt the SHErrLoc approach,  which finds a maximum likelihood explanation for static errors.