cucapra/filament #76: Checking sharing with `for` loops

rachitnigam, 2023-02-19:
If we have the following program:
```
A := new Add[32];
for #i in 0..#W {
  a := A<G+#i>(10, 20); // Uses the adder #W times
}
```

We need some way to check that the various shared uses of `a` are safe. The naive encoding is of course just unrolling the loop and adding constraints between each use. However, we can use the fact that the shared uses come from a structured loop to generate more efficient constraints. Namely:
1. To check that two invocations do not conflict, we need to enforce that `#i` >= delay of A. This is sufficient because each use of `A` is guaranteed to be `#i` cycles after the previous one.
2. To check that the parent component's delay is long enough, we can require that `max(#i)  + delay of A - min(#i)` <= delay. This is again sufficient because all uses of `A` are going to be contained within the range of `#i`.

The bigger challenge is what do when we have uses of `A` instead multiple contexts? I think the above can serve as a guideline for generating efficient constraints instead of the normal thing of O(n^2) constraints for each reuse.

gabizon103, 2023-09-05:
@rachitnigam question about (1): I don't think the loop guarantees that each use of `A` is separated from other uses by `i` cycles; the way this loop is written, each use of `A` is separated only by 1 cycle. So if `A` has a delay of 2, we would want to throw an error here because we can only use it every 2 cycles, but we would have a bunch of uses like `A<G>`, `A<G+1>`, `A<G+2>`, and so on. A correct program should have uses `A<G>`, `A<G+2>`, `A<G+4>`, and so on. In the above example, if we check `i` >= delay, we would only catch this because the loop starts at 0, which doesn't seem rigorous enough because if the loop were to start at 3 (for example), we would generate uses `A<G+3>`, `A<G+4>`, .... These uses conflict with each other but would pass the `i` >= delay constraint.

A bit lower level, but for (1) I figure that we want to add constraints saying `i` >= delay of A - but doesn't this run into the problem we talked about earlier about how `Fact`s get hoisted? I'm not familiar with the specifics of the `discharge` pass, but if we generate a bunch of constraints saying `i` >= some `d`, I have a feeling they won't ever get meaningfully discharged (with `i` having the right binding) with the current setup

rachitnigam, 2023-09-07:
Yup! You’re absolutely right about this! Some sort of structured “step” syntax seems useful because it keeps the constraints simple/tractable. I think I have an idea on how we can generalize this a bit more. Let’s schedule some time to talk?

gabizon103, 2023-09-07:
Cool! When would be a good time to talk?