cucapra/filament #122: Passing binary values in with data files

gabizon103, 2023-05-08:
Rachit and I talked about this briefly - I thought it would be convenient if we could represent inputs in binary/hex when testing components. I don't think JSON files will accept things that aren't in decimal, though.

This came to mind when I was trying to test a decoder for RISC-V instructions. To test it, I was writing out the instructions, converting it to decimal, and putting that value in the JSON file. This was a bit unwieldy and also made it less clear what each test was targeting.

rachitnigam, 2023-05-10:
The [cocotb-base harness][cocotb] exists in [this file][utils] and is invokes from a [fud stage][fud] when we run the `fud e ... --to cocotb-out` command. The data file is passes in here: https://github.com/cucapra/filament/blob/main/fud/filament.py#L204

I think the best place to add this feature is in the harness file itself. The data file is loaded here: https://github.com/cucapra/filament/blob/main/fud/filament.py#L204

We can add some sort of JSON modification that takes data that looks like a binary string and transforms it into an appropriate decimal number. Once that transformation is done, we can pass the `data` argument in the harness without any other changes. Does this make sense?

[cocotb]: https://docs.cocotb.org/en/stable/quickstart.html
[fud]: https://github.com/cucapra/filament/blob/main/fud/filament.py
[utils]: https://github.com/cucapra/filament/blob/main/fud/harness/utils.py

gabizon103, 2023-05-17:
Oops, I read this but apparently never responded. It makes sense and I'll try doing this probably next week!