cucapra/filament #125: Enumerations and methods on components

rachitnigam, 2023-05-09:
Oftentimes, components can have different sets of behavior depending on the subset of the signals being assigned to. For example, a memory will often have two different interfaces: one for reads and one for writes

```
comp Mem(addr: 8, in: 32, read_en: 1, write_en: 1) -> (out: 32);
```
This component has two interfaces:
1. `write_en` tells the component that the data on the input needs to be written to the address.
2. `read_en` tells the component that the user is trying to read data from an address.

In Filament land, we can write the following signature:
```
```

rachitnigam, 2023-05-09:
Couple of related issues:
1. Doing a good job with modules (#118) becomes even more important now since the enumerations might be tied to the definition or usage of particular components. A pleasant experience with methods requires careful thought about namespacing and various related issues
2. We need to make this work with existential quantified parameters (#121). This will allow us to implement components where we can say "the latency of read and write interface for this memory is not defined yet. We'll do some compiler magic once we have the whole design and figure out the right latency so that your design can run efficiently but the trade-off is that you need to write more abstract code". This related to how we're thinking about `virtual` operations in Calyx (https://github.com/cucapra/calyx/issues/1151 and https://github.com/cucapra/calyx/issues/1175).

sampsyo, 2023-05-10:
Drive-by comment that may be entirely obvious: in an extremely high-level way, this resembles method overloading in OO languages. That is, in overloading, the behavior (and the return type) depends on the types of the arguments at the call site. Anyway, this isn't the same because you're explicitly choosing which behavior to use at the cal site rather than using the types, but it's not completely different!

rachitnigam, 2023-05-11:
A close analogy is [Scala's match types][scala-match] which allow return types to be changed based on the call site argument. Unsurprisingly, they very quickly become useful in defining dependently typed methods.

[scala-match]: https://docs.scala-lang.org/scala3/reference/new-types/match-types.html