cucapra/filament #134: `Shift` primitive should take parameter denoting reuse

rachitnigam, 2023-05-11:
Currently, if you have to delay a signal by `N` cycles, you have two choices:
1. Use a `Register` which will constraint your II to `N-1`
2. Use the `Shift` primitive which is fully pipelined and will allow II=1

As usual, there is an entire trade-off space here: some designs need the II=1 but others might not and we should allow the programmer to express the least constrained interface, i.e., use the smallest II that would work for them.

To this end, we should redesign the `Shift` primitive to take an additional parameter that denotes what the required `II` is and generates its circuitry to support that II while exploiting it to use fewer registers. With #133, we can add a default parameter to shift registers which makes the II=1.

The redesign itself is pretty localized and simple: instantiate registers instead of `Delay` within the loop body and make them hold the signal for `N/I` cycles where `N` is the required delay and `I` is the initiation interval required.

UnsignedByte, 2023-08-15:
Could this be theoretically further simplified with the addition of #195 if we can instead define an event that is `I` times slower than the domain it is instantiated in, and just generate the original shift register instead? Just an idea, not sure if it is realistically practical or even more intuitive to understand.

rachitnigam, 2023-08-15:
Hm, I'm not sure I follow. Could you write up some pseudocode for that?