cucapra/filament #135: Consider creating a frontend CLI binary to invoke the compiler

kayagokalp, 2023-05-11:
Many compilers ship with a dedicated tooling with them nowadays. Like `cargo` with rust, `forc` with sway etc. As filament grows it will be wise to have a dedicated tooling grow with it. Since filament is a HDL, the tooling needs are not incredibly complex. To start the tooling effort, a frontend cli to the compiler can be implemented. Like `rustc` to rust. And if a dedicated tooling effort is planned, a future tooling orchestrator can call the `rustc` equivalent.

Currently filament has a `main.rs` which directly tries to compile the input source. By exposing the compiler entry points as a lib, tooling effort can be unblocked.

1. Change filament into a workspace
2. Take compiler entry points and wrap them in a crate, like `filament-core`
3. Add a binary crate to the workspace which is the CLI front-end of the compiler

rachitnigam, 2023-05-11:
Solid idea! Once again, we should take a page out the [calyx rep][calyx] and separate out Filament into several crates. #126 lays out a vision for rewriting the Filament compiler into several IRs and passes which would be ripe for this effort. A possible litmus test is being able to implement a simple [language server protocol][lsp] which can integrate with IDEs and provide features like "jump to definition" and autocompletion.

[calyx]: https://github.com/cucapra/calyx 
[lsp]: https://learn.microsoft.com/en-us/visualstudio/extensibility/language-server-protocol?view=vs-2022

rachitnigam, 2023-09-22:
This is now done

closed 2023-09-22 by rachitnigam