cucapra/filament #138: Syntactic niceties

rachitnigam, 2023-05-12:
Some syntactic sugar that might make the language easier to work with: 
- [ ] Same bound for multiple parameters: `#N, #X < 10` is the same as `#N < 10 , #X < 10`
- [ ] Fully splat a bundle. Instead of having to write `bun{0..#N}` where the bundle's size is `#N`, write `bun{..}`

gabizon103, 2023-06-02:
- [ ] Easier bit slicing, like `in[2:0]` instead of using `Slice` which can be a bit hard to read

rachitnigam, 2023-06-30:
Elide bundle type binding if the parameter is unused:
```
bundle foo[N]: [G, G+1] 32
```

In this case, all the indices in the bundle have the same availability.

UnsignedByte, 2023-10-05:
> * [ ]  Easier bit slicing, like `in[2:0]` instead of using `Slice` which can be a bit hard to read

Not sure if this should be done (yet); seems like a slippery slope to `a + b` for ports which would change the language a lot.