cucapra/filament #157: Float bus types and custom operators

UnsignedByte, 2023-05-28:
Should we think about implementing custom bus types (`float`, `int`, `uint`) with their respective primitive operators `*`, `-`, `+`, `/`? Alternatively, we could provide float operators `*.`, `+.`, etc like what is used in ocaml. The former idea could allow for more structure and explicitness in what types of data are expected and output from modules.

UnsignedByte, 2023-05-28:
Might also be more viable of an idea once structs are added.

rachitnigam, 2023-05-29:
This sounds good! BTW, there is already an implementation of floating point add and mul here: https://github.com/cucapra/filament/tree/main/evaluation/floating-point

Not sure if it follows best practices since it is directly transliterated from the Verilog but might be a good starting point.

UnsignedByte, 2023-06-28:
Two good options to deal with bus types and operations on them:
* Ignore them (treat all buses as simply a string of bits, and define custom operators on them)
* Develop a custom bus type system with defined methods, etc (this could allow for more software-like development, with tuples, structs, etc as well all with custom methods).

The latter idea would require way more work and is probably not worth considering quite yet, and would require a plethora of other language features to be implemented first.

closed 2023-06-28 by UnsignedByte