cucapra/filament #167: Explicitly printing exprs in the IR

gabizon103, 2023-06-06:
I'm trying to print exprs explicitly in the IR, for example something that currently looks like: 
`%inst0 = instance %comp0[%e2];`
should instead be:
`%inst0 = instance %comp0[32];`. @rachitnigam suggested using `Printer::expr(..)` to do this, but I ran into some questions about the structure of the existing code. 

It looks like the way instances get printed is by [this call to `index_store()`](https://github.com/cucapra/filament/blob/27bdc1326b7b8c22e72c9d848926359a8b47f813/src/ir/printer.rs#L422), which then calls [`fmt()`](https://github.com/cucapra/filament/blob/27bdc1326b7b8c22e72c9d848926359a8b47f813/src/ir/control.rs#L65)? If this is the case, then I'm not sure how to make a call to `expr()`, since printing these exprs is done outside of the printer itself. This could just be me not understanding some Rust stuff so let me know if I'm missing something here?

rachitnigam, 2023-06-06:
Yup! You'll have to write a new function that takes an `Instance` and prints out the internals. Then, you can use the `.iter()` method on index store to get a `(InstIdx, Instance)` pair and call the function to print out each instance correctly. Does that make sense?

gabizon103, 2023-06-06:
Yep, just got some of it working! Will do the rest soon. Thank you!

rachitnigam, 2023-06-09:
@gabizon103 was this addressed by #168? If so, please close the issue as well. For future PRs, you can configure them to automatically close the relevant issues using [Github's closing keywords](https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue).

gabizon103, 2023-06-09:
Oops, yes, this was addressed. Thank you for reminding me!

closed 2023-06-09 by gabizon103