cucapra/filament #173: Harness for interfacing with Calyx memories

gabizon103, 2023-06-12:
Opening an issue to keep track of this, since it is what I'm currently working on and it seems important for general Filament use as well.

My current idea is to compile a Filament design to Calyx and instantiate it, along with whatever memories it needs to interact with, in a top-level Calyx program. I'm not sure how well this approach would scale/if there are more convenient approaches (like adding standard memory components to Filament that get compiled to Calyx memories?), but I'll post updates/discussions here.

rachitnigam, 2023-06-12:
Awesome! One deeper, language-level problem here is that every port in Filament represents a "stream" of data and Filament programs transform the streams in some way. Here, we seem to want an outer "controller" that decides how to send the data for the stream. 

Maybe we can do this within the language itself? Spade uses the `entity` abstraction (which I don't deeply understand) to represent such controllers. The other question is that such controllers probably don't have the same kind of type system as Filament in general

gabizon103, 2023-06-20:
I've been working with this quite a bit and I think there is a strong argument for adding this to the language itself. It feels like you do all this work writing things in Filament so that the timing is correct, but then that sort of goes out the window when you want your design to use any memories. Would it make sense to try writing a Filament signature for the Verilog implementation of the Calyx memories?

rachitnigam, 2023-06-20:
Absolutely! You may even consider trying to implement a multi ported Verilog memory and giving it a filament. I’d be curious to see how it works and if we can do the whole darn thing in filament!