cucapra/filament #185: Writing & reading from registers in Filament

gabizon103, 2023-06-19:
I've run into this a few times while working on `frisc` things (program counter, register file), where I want to write a new value to a register while also reading the old value. Filament's current register primitive signature reflects the use case where we would want to know when we could read the new value that we are writing to the register. This is good for pipelining (and this is how it is used in most of the examples), but not for the case where I want to use whatever the active output of the register is and not wait a cycle.

The way I've gotten around this now is by writing these components in Verilog, but this seems kind of wrong and un-Filament-like? Is it worth defining a new primitive for this?

rachitnigam, 2023-06-19:
Have you looked at the Prev component in the state.fil primitives file?

gabizon103, 2023-06-26:
Haha oops

closed 2023-06-26 by gabizon103