cucapra/filament #192: Syntax change for parameter definitions and events.

UnsignedByte, 2023-06-26:
Parameters are currently much more common in expressions than events, so to clean up the syntax of filament we could remove the requirement for pound symbol in the `#PARAM` syntax and, to retain visual difference between events and parameters, add a prefix (`@` for example) to the event names in the syntax.

UnsignedByte, 2023-08-29:
This creates an ambiguity issue with `let` bindings: `let PARAM = PARAM2` now has no syntactic difference from `let port = port2`. Should we change `let port = port2` to `let port p= p2` or something like that? @rachitnigam

rachitnigam, 2023-08-30:
No, this is why it’s a non trivial change because it requires syntax disambiguating to figure out what is a port, a param, and an event.

rachitnigam, 2023-08-30:
Let me expand on the comment above a little bit. This *ambiguity* problem shows up almost every single language. For example, in Rust, type identifier, struct names, and variables all use the same syntax:
```rust
struct Foo;
type Bar = Foo;
fn baz(x: Bar) {
  let Bar = x;
}
```

The problem for the rust frontend is to look at this code and figure out the correct binding for each symbol. In this case, the *position* of each identifier exposes the binding chain we need to look at. When you look at `type <ID> = <Expr>`, the `Expr` can only mention other identifiers in the type name bindings. Similarly, when looking at `let <ID> = <Expr>`, we know that `Expr` comes from the identifiers defined as variables.

The reason Rust doesn't have to do what Filament does, which is have distinct looking identifiers for different kinds of terms, is because it separates the frontend from the core representation(s). Now that we also have that separation in Filament, what we should actually do is remove the syntactic difference between ports, params, and events completely and use `astconv` to resolve these things for us.

rachitnigam, 2023-10-20:
We've made this change!

closed 2023-10-20 by rachitnigam