cucapra/filament #195: Clock and reset domains

rachitnigam, 2023-06-27:
The clock and reset domains in Filament are currently implicit: each event is implicitly tied to the global clock signal and uses the implicit `clk` signal. Similarly, all state elements, which contain registers at their leaves need a `reset` signal (since they implement synchronous resets) which is implicitly threaded through the design by Calyx.

However, there are a couple of good reasons to support explicit modeling of these signals (with good inference algorithms) because any serious design is going to require careful reasoning of these signals. Furthermore, it'll let us support cool features like:
* Automatic clock gating logic generation (suggested by @jrk)
* Multi-pumped memories (for example, running the memories as twice the frequency of the compute)
* Checking reset consistency (little ill formed right now but something that'll tell the user that parts of their design have not been correctly reset)

## Clock Domains
Continuing to pay homage to Rust, we can adopt the lifetime syntax to define the available clock domains in a component:
```
comp Foo[W]<'c1, 'c2, G: 'c1, L: 'c2>(...)
```

Each event is explicitly tied to a particular clock domain and events from different domains are not allowed to reuse instances or interact with each other. However, if we can prove that the clock domains have *some* relationship to each other, then we can use timing information to implicitly synchronize them. This might require some magic external component definition that can convert things from domain `c1` to `n*c1` where `c2` pulses `n` times faster than `c1`. A cool possibility here is that `n` is just parameter to this component and can therefore describe arbitrary domain synchronizers:
```
comp Sync[N]<'c1, 'c2=N*'c1, G1, G2>(
  in: [G2, G2+1] 32
) -> (
  out[N]: for<_> [G1, G1+1] 32
)
```

Note that in moving values from the fast domain to the slow one, we get `N` values on the output which is exactly what we should expect.

## Reset Domains

Reset domains are easier mostly because I'm not sure what kind of optimizations or automatic code generation people would care about in this case. The important thing, however, is explicitly adding reset domains to state elements and checking that components that use them themselves have a reset domain available.

rachitnigam, 2023-06-27:
Clocks and parameters should probably go in the parameter list instead of the event list since, once a circuit is instantiated, it is only allowed to have one clock or reset signal (is that true?)

sampsyo, 2023-06-27:
Cool; this sounds like the essence of the matter:

> Each event is explicitly tied to a particular clock domain and events from different domains are not allowed to reuse instances or interact with each other. However, if we can prove that the clock [...]

That is, if you invoke a single instance from one clock domain, by default you shouldn't be able to use it from any other clock domains, because we lose the ability to prove mutual exclusion. But there may be some other way (not a current Filament feature) to show that you have waited long enough and you can be sure no other invocations from other clock domains are using something anymore, which would make this possible again.

Clock domains that are multiples of each other are one such option, and another is some kind of dynamic clock synchronizer thingy.

> Clocks and parameters should probably go in the parameter list instead of the event list since, once a circuit is instantiated, it is only allowed to have one clock or reset signal (is that true?)

Not sure about reset signals, but I think it makes sense for clock parameters to be at the type/event level (as in your example) rather than parameters? Otherwise, I don't quite see how you get to a point where a single hardware structure combines two different clock domains, which is where these interesting interactions happen.

rachitnigam, 2023-06-28:
Oh interesting! I was thinking of synchronization of values between circuits running in different clock domains but you're talking about the very Filament-esque idea of safety of reuse among different clock domains. I think that might be cool to think about but not sure if anyone does it in practice.