cucapra/filament #199: Floating point library bug somewhere

UnsignedByte, 2023-07-02:
The floating point module has a rare bug leading to different behavior from expectation. This could either be due to the 5 shift checks in the other 3 libraries or some other reason.

Full trace (copied from a run of runt in workflow) shown [here](https://github.com/cucapra/filament/actions/runs/5436020310/jobs/9885576606#step:7).

rachitnigam, 2023-07-02:
GitHub actions logs are not retained so let’s get the minimal failing run in this issue.

rachitnigam, 2023-07-08:
@UnsignedByte I'm going to disable the testing of your FP library since its broken and it's stopping other PRs from being merged. Please fix it soon and we can re-enable it.

rachitnigam, 2023-07-08:
Here is the minimal reproducible example. The data file is:
```json
{
    "left": [
        479189982
    ],
    "right": [
        2626205011
    ]
}
```

And the command to run (after saving the file to `data.json`) is:
```sh
fud e --to cocotb-out -s cocotb.data data.json -s futil.flags ' -d canonicalize' evaluation/floating-point/add/harness.fil | jq
```

The generated output is:
```json
{
  "out": {
    "0": [
      478267811
    ]
  },
  "verilog_pipe": {
    "0": [
      478267811
    ]
  },
  "verilog_nopipe": {
    "0": [
      478267811
    ]
  },
  "filament_lib": {
    "0": [
      442814848
    ]
  },
  "cycles": 5
}
```

rachitnigam, 2023-07-09:
I'm not sure if we should take the current FP implementation as the source of truth. One possible way to figure out if we're doing things correctly is using something like [TestFloat](http://www.jhauser.us/arithmetic/TestFloat.html) library. Checking the correctness of floating point in general is a morass of complications so maybe the right approach is using Berkeley's [HardFloat](https://github.com/ucb-bar/berkeley-hardfloat) library and move the correctness claims to them.

UnsignedByte, 2023-07-14:
> I'm not sure if we should take the current FP implementation as the source of truth. One possible way to figure out if we're doing things correctly is using something like [...]

Confirmation from C++: the current FP implementations are incorrect, the filament library matches a C++ implementing `iec559` floats.

rachitnigam, 2023-07-14:
Ha! Thanks for figuring this out! Can you also provide the C++ you used to confirm this so we have this in the future. And yeah, we should remove the existing implementations for this reason.

UnsignedByte, 2023-07-14:
The C++ code is here:
```
#include <cstdio>
#include <cstdint>
#include <limits>

int main() {
    printf("%d\n", std::numeric_limits<float>::is_iec559);
    uint32_t a = 479189982;
    uint32_t b = 2626205011;

    printf("%x + %x\n", a, b);

    float aa = *((float*)(&a));
    float bb = *((float*)(&b));

    float res = aa+bb;

    printf("%g + %g = %g\n", aa, bb, res);

    uint32_t rres = *(uint32_t*)(&res);

    printf("%x, %x, %x\n", rres, 478267811, 442814848);
}
```

UnsignedByte, 2023-09-22:
Closing this because it turned out to not be a bug in the filament library, and instead in the other implementations.

closed 2023-09-22 by UnsignedByte