cucapra/filament #200: IR passes to implement

rachitnigam, 2023-07-02:
The old compiler flow has a bunch of different passes that need to be transitioned before we replace it with the IR:
- [ ] Add assumptions
- [ ] Type check
- [ ] Interval check
- [ ] Sharing check
- [ ] Phantom check
- [ ] Monomorphize
- [ ] Bundle elimination
- [ ] Lower
- [ ] Compile to Calyx

rachitnigam, 2023-07-02:
Not sure if I've mentioned this before but the following command will print out the program AST after each pass in the existing compiler flow:
```
cargo run -- <file> --log debug
```

Use this to check whether a particular pass generates the right kind of code

gabizon103, 2023-09-05:
This was accidentally closed when `ir-dev` got merged into `main`, but the sharing check still isn't implemented so I'm reopening until that's done