cucapra/filament #212: Storing port names of signature ports in main and primitive components.

UnsignedByte, 2023-07-10:
Currently all original information about the physical program (names for ports, invokes, components, etc, and locations in the program they are bound) is stored in `Info` structs, and because `Info` is meant to be a struct that should technically be removable without breaking compilation, we need to store the true signature port names for a component somewhere to be used during compilation. 

This is necessary because verilog blackbox externals need the right port names in order to properly connect, and similarly the main component should keep this information in order for the user to be able to properly interface with the program. 

This information could be discarded for other user-defined components, instead relying only on port IDs to generate names (#211) so the question becomes how this information should be stored in the component.

UnsignedByte, 2023-07-14:
Could add an optional boxed datastructure storing all the original necessary information for the external interface (signature port names, etc) and have the existence of this datastructure be equivalent to a flag for whether to preserve this information on compile, necessary for toplevel components and externals.

rachitnigam, 2023-07-26:
This is now done in the `ir::Component` right @UnsignedByte?

UnsignedByte, 2023-08-04:
This was resolved by #186 which added `InterfaceSrc`.

closed 2023-08-04 by UnsignedByte