cucapra/filament #226: Astconv fails on recursive and mutually recursive components

UnsignedByte, 2023-07-20:
Astconv currently uses `Traversal`  to generate `Foreign` ports properly, but this method will not work for recursive and mutually recursive components, as they have dependencies on one another. The issue is that the signature ports and events have to be compiled for all dependencies before the commands of the component itself can be compiled, which doesnt work when there are circular dependencies.

An alternative solution is to compile all signature ports and events in a `declare_comp` pass for all components before going through commands (and invokes). This would require declaring and storing all signature information in some mapping before converting each component.

rachitnigam, 2023-07-20:
Wait, it also fails on self-recursive programs? We need to fix that because the tests on `main` use `ReduceAdd` which is self recursive

UnsignedByte, 2023-07-20:
Currently working on solving this, but this might end up requiring quite a significant rewrite of the astconv pass.

Summary of the problems:
* Need to be able to store the signatures of every component (containing portidxs, etc) somewhere so that ctx can remain immutable during conversion
* Also need to pass over all components to convert their signatures first, before converting commands

We can solve these by storing input and output port idxs somewhere in an external mapping and then using these to generate foreigns, but it doesn't change the fact that we will need a second pass over the components.

rachitnigam, 2023-07-20:
Yeah, this going to be tricky. Let’s punt on it now as long as we can get self recursion working and address it after IR flow Is complete.

UnsignedByte, 2023-07-20:
> Yeah, this going to be tricky. Let’s punt on it now as long as we can get self recursion working and address it after IR flow Is complete.

Self recursion has the same issue, as it would require foreigns of itself, meaning it would have to have its signature compiled in the context which currently doesn't happen as it is `std::mem::take`n from the context before compilation. Either we `clone` the component instead or we would have to implement one of the above scenarios.

closed 2023-08-04 by UnsignedByte