cucapra/filament #233: IR printer changes

gabizon103, 2023-07-25:
Comments about things that were removed in #231, but were kind of nice:
1) It was nice to see param owners printed out with params, so I could verify that the ports/params matched up
2) Printing out ports using the source-level names instead of the IR idxs is nice, but when unrolling loops I use the same info as whatever was in the loop body for all the things I generate, so it kind of looks like the same port is being reused when that isn't happening. It would be nice to also have the IR idx somewhere in there.

@rachitnigam if these changes are cool with you, I'll implement them

rachitnigam, 2023-07-25:
Yeah! One thing we can do is implement this when `log::debug` is enabled (see printing for `ExprIdx`). That way, the default printing mode is "simple" while debug mode prints out more verbose information. Would that work?

rachitnigam, 2023-07-25:
And yeah, for (2) I was going to say that this might be a good reason *not to copy* the `Info` object since those are semantically distinct ports and shouldn't share a name.

gabizon103, 2023-07-25:
Yeah that was another question I had - only thing I was iffy about is I'd have to be making up new `ast::Id`s that don't actually exist in the source program

rachitnigam, 2023-07-25:
Yeah, but technically there shouldn't be any source-level error messages after monomorphization any ways

rachitnigam, 2023-07-25:
Also, the `Info::Empty` will probably break a bunch of code since we do: 
```let Info::Event { ... } = event.info else { unreachable!(...) }``` 
In a bunch of places in the code.

We should switch to having a different interface for destructuring `Info` in the future so this stops being a problem

UnsignedByte, 2023-07-27:
#239 will probably affect this change, FYI.

rachitnigam, 2023-08-07:
@gabizon103 has these been addressed?

gabizon103, 2023-08-07:
Oops, yeah this was added in #216

closed 2023-08-07 by rachitnigam