cucapra/filament #236: Expressions with operations on concretes should be interned simplified

UnsignedByte, 2023-07-26:
We might want to simplify when interning for operations like `add`, `pow2`, etc between two constants, because currently these are done in the `ExprIdx` implementations of `add`, `pow2` helper functions but these aren't enforced to be used everywhere (such as in the old `Subst` code, fixed in #235), leading to non-concrete expressions that are actually fully concrete. Moving this optimization into `intern` itself would simply save some headache on that part, but this might need to be discussed.

rachitnigam, 2023-07-26:
Yeah, that seems reasonable to me but it's somewhat tricky because the `Component::add` implementation for `Expr` adds things into the store in an unsimplified manner. If we try to change it (#220), the stack overflows because the `ExprIdx` "smart" constructors need to call `Component::add` as the back-off case.

There is a `Simplify` pass that attempts some simplification but that's somewhat different from this issue.

UnsignedByte, 2023-07-26:
Yes but I was thinking we could just move the "smart"ness of the `ExprIdx` constructors directly into the component add, so infinite recursion should not become an issue.  In fact, recursion should not be a problem at all because all already-interned expressions will be simplified so should need to be no recursive call.

rachitnigam, 2023-07-26:
Sure! I'm not opposed to the change!

closed 2023-07-27 by UnsignedByte