cucapra/filament #261: Implement `Ctx` for `Context` for resolving `Foreign`

rachitnigam, 2023-08-10:
The `ir::Context` structure is the only thing that can resolve `Foreign` indices since we need both the component and the underlying index type. Providing convenience `get` methods for all the various `Foreign` in the IR would be useful. However, the current `Ctx` trait is not amenable to this change since we need to implement `add` as well. It might be useful to separate the trait out so that we can implement `get` methods.

rachitnigam, 2023-08-25:
We should change the `Ctx` stuff to use a `IdxLike` instead of an `Idx`, implement `IdxLike` for `Foreign`, and use that to implement `Ctx` methods for `Context`.

UnsignedByte, 2023-08-29:
Could we not have the add function insert the held key into the relevant component as well? Maybe this would be weird behavior, I'm not sure.

UnsignedByte, 2023-08-29:
Nevermind, I forgot `add` only takes `T` so would not specify what component to insert into.

PR #288 (2023-08-30): Implement Ctx for Foreigns.
Fixes #261 and also generalizes some existing functions to use `Ctx` instead of `Component`
closed 2023-09-01 by rachitnigam