cucapra/filament #284: Debug compilation flag

UnsignedByte, 2023-08-24:
Would be helpful to implement something like rusts's `cargo build --release` for filament as it would be nice to preserve component/port names in the compiler if a flag is enabled and the info is present, for example when debugging through gtkwave or through looking at the resultant calyx/verilog code.

UnsignedByte, 2023-08-24:
@rachitnigam I'll go ahead and implement this in compile as I think it is a useful feature nonetheless, but if this would be better enabled in a different way we could look into that as well.

rachitnigam, 2023-08-25:
Yeah! I think this is a good idea. I was going to mention that we should probably try to preserve as many source-level names as possible anyways (especially with component signatures). Storing and using this information in the `ir::Info` seems feasible to me (in fact, it already is stored there) but let me know if you had a different design in mind.

closed 2023-09-01 by UnsignedByte