cucapra/filament #299: Attributes for components

rachitnigam, 2023-09-01:
Going forward, it would be useful to expose some mechanism of adding attributes to components (and other things like instances) to control code generation. For example, we can put an attribute to force generation of an FSM with a particular delay:

```
#[slow_fsm=5]
comp main<G:10>(...) {...}
```

The attribute can then be used by the lowering pass to generate the FSM of the specified delay (and circumvent the need for #273 for some time).

UnsignedByte, 2024-08-27:
A quick question on attributes before we implement them should be about scoping of different attributes. For example, say I have a two-event component as follows:
```rust
#[slow_fsm=5]
comp main<G: 10, L: 9>(...) -> (...) {...}
```
which is currently not super realistic to create currently but nonetheless exists, and could become important in dynamic filament. A component level attribute like `slow_fsm` would not be able to distinguish between the two events and which one to apply the slow FSM to.

Even further we may have attributes we want at the port level, such as information about how bundles could be lowered at the top level. The simplest example of this is with the spiral DFTs, where the verilog emitted can have say, `4` input ports for a `16` point FFT where the first input represents the `0th, 4th, 8th, 12th` input in each cycle. We have discussed before possibly having some form of lowering where a bundle like
```
A[4]: for<i> ['G+i*4, 'G+i*4+1] 32
```
would be lowered into a single port that is reused at different times for different behaviors. This behavior could be decided via attribute, where `#[collapse_bundle]` or something similar could decide whether a bundle is collapsed into one port or split into multiple.

UnsignedByte, 2024-08-27:
Of course component level attributes could come first but I do think this is an idea that might be worth exploring in the future.

rachitnigam, 2024-09-02:
I think the approach you've taken---to implement the simplest thing first (component-level attributes)---is probably the right one. As we add more features to the language (real modules, real scoping rules), we can investigate more fine-grained attributes.

closed 2024-09-04 by UnsignedByte