cucapra/filament #301: Update Filament docs

gabizon103, 2023-09-01:
On the [Getting Started page](https://filamenthdl.com/docs/start.html) it says to configure `fud` with `cargo build && fud config stages.futil.exec "$(pwd)/target/debug/futil"`. This should be changed to `cargo build && fud config stages.calyx.exec "$(pwd)/target/debug/calyx"`

Also, syntax highlighting is not mentioned, but should probably be included. For VS Code:
```
cd $HOME/.vscode/extensions
ln -s <filament root directory>/tools/vscode filament.filament-0.0.1
```

rachitnigam, 2023-09-02:
Thanks for finding this! In fact, the `calyx` configuration is probably no longer needed because we invoke the calyx compiler ourselves but reasonable to update it anyways!

CrepeGoat, 2024-08-24:
related?: the "getting started" page says to run `cargo run -- tests/compile/par.fil` to test the build.

I get a few errors:
```sh
$ cargo run -- ./tests/compile/par.fil
error: `cargo run` could not determine which binary to run. Use the `--bin` option to specify a binary, or the `default-run` manifest key.
available binaries: fil-gen, filament
$ cargo run --bin=filament -- ./tests/compile/par.fil
warning: ...
warning: ...

Error: invalid file: Failed to read ./tests/compile/par.fil: No such file or directory (os error 2): 
Compilation failed with 1 errors.
Run with --show-models to generate assignments for failing constraints.
```

I can't find a file named `./tests/compile/par.fil` either. so it seems that the docs should:
1. include the `--bin=filament` flag
2. list a working test file

CrepeGoat, 2024-09-02:
> related?: the "getting started" page says to run `cargo run -- tests/compile/par.fil` to test the build.
> 
> I get a few errors:
> ...

moved ^that request to a new issue: #451