cucapra/filament #308: Functions on parameters using existential parameters

UnsignedByte, 2023-09-04:
@rachitnigam and I discussed this in a meeting, but the addition of existential parameters by #121 enables creating components used only for operating on parameters.

For example, we can now do the following to implement `pow2`:
```
comp Pow2[N]<'G: 1>() -> () with {
    exists K;
} where K > 0, N >= 0 {
    if N == 0 {
        exists K = 1;
    } else {
        P := new Pow2[N-1]<'G>();
        exists K = P.K;
    }
}
```

This is a pretty cool observation as it allows us to do arbitrarily complex computation on parameters without making the compiler any more complicated, but still has some caveats:
* The assumptions necessary for type checking are currently not added using this, so such an implementation cannot be used somewhere that needs to be validated at that point in compilation (I.E. `ReduceAdd` or equivalent).
* These functions are components, so we cannot use them simply as functions on parameters, and can only define them in component bodies (though the existential parameters can likely be accessed outside of that scope, @rachitnigam correct me if I am wrong here).

This also allows us to use the constraints on `K` to generate assumptions in our parent component, which is also a nice observation (although `assume` already exists, so this may be less important). This is not really an issue, but just recording some of the possibilities with existential parameters, and maybe we could add some syntactic sugaring to this to make some cool features in the future.

rachitnigam, 2023-09-05:
The code in the example is not quite right. If we want to compute K = 2^N, then we have to do:
```
comp Pow2[N]<'G: 1>() -> () with {
    exists K where K > 0;
} N >= 0 {
    if N == 0 {
        exists K = 1;
    } else {
        P := new Pow2[N-1]<'G>();
        exists K = 2 * P.K;
    }
}
```

Apart from that, yeah, this is a cool capability. Essentially, forall parameters act as inputs to components and existentials are "outputs". The notable thing about this component is that it doesn't actually do anything at runtime; it's more like a "type-level" function whose only goal is to produce a new type based on a given type.

The reason this encoding is interesting (and tractable) is because while the computations on parameters can be arbitrary, the type system doesn't know about it. That is, even though, mathematically, we have the constraint that `K == 2^N`, the only fact exposed to the type system is that `K > 0` so users of this parameter can only rely on this knowledge.

rachitnigam, 2023-09-05:
A relevant reading here is [liquid types](https://goto.ucsd.edu/~ucsdpl-blog/liquidtypes/2015/09/19/liquid-types/) which also allow for reasoning of predicates over types in a decidable manner. Maybe one way to decide which functions (like `pow2` and `log2`) make into the core language is keeping the language fragment decidable.

rachitnigam, 2023-09-22:
Is there anything actionable with this issue?

rachitnigam, 2023-10-19:
Fstar has a similar idea with the `Lemma` return type that they use to describe libraries of helpful axioms: https://github.com/FStarLang/FStar/blob/master/ulib/FStar.Math.Lemmas.fst#L66