cucapra/filament #309: Parameter Bundles/Structs with constraints

UnsignedByte, 2023-09-04:
This was discussed in a meeting with @rachitnigam, but often times similar functions require very similar parameters. A simple example is parametrized floating point, where all floating point operations take `W, Mantissa, Exponent` for the number of  `Mantissa` and `Exponent` bits, with `W = 1 + Mantissa + Exponent`. The necessity of these parameters also moves upward through any components using these modules that want to keep this parameterization capability, which results in toplevel components of larger blocks such as `FFT`s having tons of parameters that are mostly only passed to constituent instantiations. I am not sure exactly the syntax for these, but we could define them somewhat like the following:

```rust
type Float[W, Mantissa, Exponent] where
    Mantissa > 0,
    Exponent > 0,
    W == 1 + Exponent + Mantissa;
    
comp FAdd[T: Float]<'G: 1>(
    left: ['G, 'G+1] T.W,
    right: ['G, 'G+1] T.W,
) -> (
    out: ['G, 'G+1] T.W,
) {
    // ...
}

comp main<'G: 1>(
    left: ['G, 'G+1] 32,
    right: ['G, 'G+1] 32,
) -> (
    out: ['G, 'G+1] 32,
) {
    let T := Float[32, 8, 23];
    Add := new FAdd[T]<'G>(left, right);
    out = Add.out;
}
```

I chose to use `let` here to bind floats to match `let Param = ` syntax, and `:=` because we are defining a type (sort of), but this should probably be discussed.

rachitnigam, 2023-09-05:
I like this proposal! A minor nit: I would probably rename `type` to `param` or something to indicate that it's a parameter definition. It seems we have implicitly named fields for these parameter definitions which seems reasonable to me.

The `let` binding thing is tricky because the compiler currently inlines all parameter bindings but again, we need some way to name constructed values of `param` definitions. A good next step would be figuring out what changes are needed in the compiler to support. My guess would be that this abstraction could be eliminated before we hit the IR. However, `astconv` is already pretty complex and this would be extra complexity. We might have to bring back the ability to do source-to-source transformation of the AST to keep some of this tractable.

UnsignedByte, 2023-09-05:
Yeah, the `type` thing was very temporary and I wasn't sure exactly what we wanted it to be, `param` sounds great.

I do think we could probably eliminate this pre-IR, in fact we can probably inline things pretty similarly to how we already do with let bindings. The only thing would be that we would have to add the constraints to every component using the types for type checking to properly work (assertions when defined, assumptions when used in signature). Maybe we could add back some of the ast pass capability, and have something like `let` elimination or similar?

rachitnigam, 2023-09-05:
Yeah, that's what I was thinking as well. It is starting to mirror the [structure of the Rust compiler](https://rustc-dev-guide.rust-lang.org/overview.html#what-the-compiler-does-to-your-code) where there are passes that run before the core representation (MIR, in Rust's case).