cucapra/filament #316: Primitive `Slice` used ambiguously

UnsignedByte, 2023-09-08:
Slice is implemented as follows:
```v
module Slice #(
  parameter IN_WIDTH = 32,
  parameter MSB = 31,
  parameter LSB = 0,
  parameter OUT_WIDTH = 32
) (
  input wire logic [IN_WIDTH-1:0] in,
  output wire logic [OUT_WIDTH-1:0] out
);
  assign out = in[MSB:LSB];
endmodule
```
which makes `MSB` inclusive, and thus we theoretically want `OUT_WIDTH = MSB - LSB + 1` as a constraint. However, within `primitives/math.fil` component `Id_Init` declares a 
```rust
q_slice := new Slice[Ext, W, 0, W]<'G>(c_ext.out);
``` which would violate this constraint (I'm assuming `W` is meant to be non-inclusive here).

Which should we choose to use? It might be worth doing non-inclusive slicing as it would match with bundle indexing/range generation, etc.

rachitnigam, 2023-09-09:
Ah, good catch! I don't think it matters too much which one we pick as long as the constraints are consistent. Maybe we can look at Chisel, PyMTL, etc. and pick one that seems consistent with them?

UnsignedByte, 2023-09-09:
I'll use the custom of inclusive `MSB` for now, then.

PR #317 (2023-09-10): Expands the primitives library a little
Adds `ConcatBundle` and `SplitWire`, combinational components that combine bundles into a single wire and to split a single wire into a bundle.

Also adds `FNeg` and `FSub`, components to negate and subtract floating point numbers.

Will fix #316, which should remove some ambiguity in the way we use `Slice` at the moment.
closed 2023-09-09 by UnsignedByte