cucapra/filament #318: Built-in `sin` and `cos` functions

UnsignedByte, 2023-09-09:
We should add `sin` and `cos` to the builtin functions `log2` and `pow2` (mainly in this case because I need to generate the `N` roots of unity for an `N` point FFT), which I store currently using constant wires. Alternatively, I could get a hardware implementation of `sin` or `cos`, but that seems a lot more involved (and would require us to have a constant `PI`, pushing the ball down the line).

This does have some caveats however: what format should the return value be in? We are likely only going to be able to return either `Float32` or `Float64`, but that seems weird as parameters currently can only store integers, so we'd have to convert their value to a `32` or `64` bit int, which is not the most intuitive to understand or work with.

rachitnigam, 2023-09-09:
Thanks for getting started on this! One important thing to mention here is that `log2` and `pow2` are *type-level* functions which means that can be used in availability expressions. I'm assuming that's not something that you want to support here and instead just want a compile-time function to apply to parameter values.

In general, we seem to be hitting the same points as the discussion in https://github.com/cucapra/filament/issues/308 where we want a broader class of compile-time capabilities without affecting the type system. It's worth mentioning that this is one the reasons people like embedding their HDL in other software languages (like Chisel and PyMTL); this kind of stuff becomes trivial.

To make progress on this, I think we need to define some clear goals for what the library of compile-time/parameter-only functions are allowed to do and how we want to use them going forward. Once we have that, it should be easy to implement.

FWIW, adding these functions to the `expr` stuff isn't necessarily bad: like `pow2` and `log2`, we could encode them as uninterpreted functions which will render them useless for any timeline reasoning.

UnsignedByte, 2023-09-09:
Yeah, I was also thinking maybe we could do something like copying over some of verilog's built in math functions [listed here](https://www.chipverify.com/verilog/verilog-math-functions) for example, or maybe even doing something like verilog's C-style macros but with more functionality (maybe even some syntax like `fn` where we can just write a rust-like function inside? I think #308 could be used as the backbone for some of the internal representation of these functions but it would definitely need to be sugared up a lot.

rachitnigam, 2023-09-10:
That list is a good starting point it seems! Although quite weird that trigonometric functions are also exposed.

As a medium term, bandage solution, we could consider implementing all of them and not adding any axioms so that they cannot really be used in liveness of ports.

UnsignedByte, 2023-09-14:
Getting started on working on this now, but in terms of using functions in the liveness of ports, I think it might be worth considering having some flag in custom functions allowing or disallowing them from being used in liveness, so that we can have better error messages for functions that don't contain axioms.

Functions like `sin` and `cos` that return pretty much meaningless integer values (as they are meant to be interpreted as floats) probably should never be used in liveness calculations anyway, as they would be really difficult to axiomize.

I can make a quick change to the `assume` pass so that functions that don't generate assumptions instead generate an error message about how they shouldn't be used in liveness situations if that works as a solution.

rachitnigam, 2023-09-14:
Yeah, that sounds like a good idea! I would go as far as saying that there should be no flag that allows you to have `sin` and `cos` in liveness. Also, let's put this in a `well-formed` pass where we can collect other checks like #322.