cucapra/filament #329: Inconsistent behavior with Iterative Divider

UnsignedByte, 2023-09-19:
The `verilog` implementation of the iterative divider returns `0` upon the division `0/0`, while all the other implementations return `255` (max value).

Placing this test in `edge.json` (provided #328 is merged) will show off this behavior.

UnsignedByte, 2023-09-19:
Looking into the verilog implementation shows that this is caused by an optimization that checks whether the dividend is `0`. Because this is
* Not a necessary optimization, as behavior works fine either way
* Changes the behavior of `0/0` in a way that is inconsistent with all other implementations
* And, most importantly, **Makes the component non-pipelineable** (as it would return the value in `1` cycle, instead of `W`)

I have opted to remove the optimization altogether.

rachitnigam, 2023-09-19:
Well, division by 0 is undefined so returning anything seems fine to me. The right thing to do is to assert an error signal to tell the user that something went wrong. If we had Spade like enums, we could wrap the return value in `Option<T>` and say it is `None` when division by zero happens.

Regardless, I think not testing for it seems fine. For other solutions, you could push the value `0` across the pipeline for `W` cycles to keep the pipelined behavior.

PR #328 (2023-09-20): Fixes bugs in FP-Add and Iter-Div and add Edge Test cases
Fixes a bug in renormalization when all mantissa bits are `0`.

Also fixes inconsistent behavior between verilog and filament implementations with the iterative divider when performing `0/0`, and a bug that made the verilog implementation non-pipelineable. (Closes #329)

Also adds testing capability for edge cases with FP library and iterative divider.
closed 2023-09-20 by UnsignedByte