cucapra/filament #330: CORDIC in Filament

gabizon103, 2023-09-19:
We want to implement more designs in Filament (ideas listed in #124), so I'll take a shot at CORDIC. Will list some resources/other implementations here

rachitnigam, 2023-09-19:
Extremely hype! I think it would be cool to spend some time thinking what kinds of design trade-offs we'd like to expose using parameters and what capabilities they can express. Once we have that, let's write up a proposal for implementation and get started!

gabizon103, 2023-09-19:
Very true, that is a good idea. I think I'll start with writing a simple unparameterized version of it to get an idea of what makes sense to expose/parameterize. I can write up a proposal after that. Does that sound good?

rachitnigam, 2023-09-19:
Sure! The pre-reading suggestion was because it might be good to understand *if* there are any opportunities for parameterization in the design or not. If CORDIC doesn't have any opportunities, then we might prefer something else for the sake of more interesting exploration.

gabizon103, 2023-09-20:
Ah, good point. I think it would be cool to do something that could show off existential parameters - I'll see if they can be applied here

rachitnigam, 2023-10-07:
@gabizon103 are you working on #332 instead of this? If so, let's close this issue for now

rachitnigam, 2023-10-12:
Okay, closing this for now but @gabizon103 feel free to reopen if you end up working on it.

closed 2023-10-12 by rachitnigam