cucapra/filament #341: Confusing error messages when bundles of length `> 1` are assigned without splatting.

UnsignedByte, 2023-10-01:
Bundles accessed without `splat` are assumed to be bundles of length `1` and elaborated as `bundle{0}`. This is not expected behavior when the bundles are not of length `1` and can lead to confusing error messages. We should explicitly disallow this syntax for non-port bundles.

The following short example shows off this behavior:
```rust
comp Test<'G: 1>(
    in[2]: ['G, 'G+1] 1,
) -> (
    out[2]: ['G, 'G+1] 1,
) {
    out{0..2} = in{0..2};
}

comp main<'G: 1>(
    in: ['G, 'G+1] 1,
) -> (
    out0: ['G, 'G+1] 1,
    out1: ['G, 'G+1] 1,
) {
    bundle pair[2]: ['G, 'G+1] 1;
    pair{0} = in;
    pair{1} = in;

    // Forgetting to splat pair using pair{0..2} here.
    test := new Test<'G>(pair);

    out0 = test.out{0};
    out1 = test.out{1};
}
```

This has the following error message:
```rust
[WARN ] A false proposition was created
error: required bundle of size `2' but found bundle of size `1'
   ┌─ out.fil:2:5
   │
 2 │     in[2]: ['G, 'G+1] 1,
   │     ^^ length of bundle is 2
   ·
19 │     test := new Test<'G>(pair);
   │                          ---- length of bundle is 1

Compilation failed with 1 errors.
Run with --show-models to generate assignments for failing constraints.
```

UnsignedByte, 2023-10-01:
FWIW, another possible solution is to change the default behavior of no splat to be `{0..len}` which seems more intuitive anyways.

rachitnigam, 2023-10-02:
Yeah, that change would also make it work in the case when the bundle has size of exactly `1`. The problem is that we need the information about the length of the bundle which is not available during `astconv` which is why we end up with this default behavior.

Getting the better default behavior requires us to redesign the frontend to support these kinds of transformations.