cucapra/filament #351: Multidimensional Bundles

UnsignedByte, 2023-10-12:
As the title says, we should have support for multidimensional bundles. I'm thinking something like this:
```
bundle a[10][20] for<i, j> ['G, 'G + i + j + 1] 32;
```
where `i` ranges from `0` to `9` and `j` from `0` to `19`.

rachitnigam, 2023-10-13:
Yeah, this is a common problem that I've run into as well, for example with the reshape operators. The big question for me is how to represent this in the IR. Do we add these to the IR or do we eliminate them beforehand. If we eliminate them, then we should figure out what `assert` statements we need to add to make the checks succeed (which is another benefit of using multi-dimensional accesses).

rachitnigam, 2023-10-14:
Let's document some more complex use cases for multi-dimensional bundles from the existing designs.

rachitnigam, 2023-10-14:
The pipelined FFT:
```
bundle data[(NStages+1) * NPoints * 2]: for<i> ['G+i/(NPoints*2) * L, 'G + i/(NPoints*2) * L + 1] W;
```

To make it more clear what's going on here, we can rewrite using `R = NStages + 1` and `C = NPoints * 2`:
```
bundle data[R * C]: for<i> ['G + i/C * L, ...]
```
We're essentially saying that everything in a particular column has the same latency and going between rows increments the latency by `L`.

Which can be rewritten:
```
bundle data[R][C]: for<i, _> ['G + i*L, 'G + i*L+1]
```
Because we have access to two distinct iterators, we can say that only changing the row changes the latency.

A big win from this is that we get rid of the `/` operator from the liveness of the bundle which is important in making the assertions scale better (as seen in #348).

PR #379 (2023-10-20): Support multi-dimensional bundles
Fixes #351. Adds supports for multi-dimensional bundles. For now, just testing to make sure that the changes don't break any reasoning with single dimensional bundles. Once that works, I'll add support to the frontend for parsing multi-dimensional bundles.
closed 2023-10-20 by rachitnigam