cucapra/filament #354: Pure function interpretation of `comp` definitions over parameters

rachitnigam, 2023-10-14:
Consider the following program:
```
comp Foo[I](...) with {
  exists O;
} { ... }
comp Bar(...) {
  F0 = Foo[10];
  F1 = Foo[10];
  F2 = Foo[20];

  assert F0::O == F1::O;
  assert F0::O == F2::O;
}
```

The question which `assert` statements, if any, should be allowed to pass. Here are the possibilities:
1. **No `assert` passes**: This means that *all instances*, regardless of input parameters, can produce different values for existential parameters.
2. **The first `assert` passes**: This means that components are *pure functions* over parameters; the only thing that can influence the value of a parameter is input parameters. The body can of course use existential parameters defined by other subcomponents but those themselves can only be influenced by input parameters.
3. **Both `assert`s pass**: This means that there must have been some language-level mechanism to declare that the existential parameter `O` has the same value for all instances of the module. In this case, the parameter *must be a constant* value and a more appropriate thing to say would be that existential parameter is defined over the component definition instead of an instance (and use the syntax `Foo::O` instead of `F0::O`).

I think the current status, which is (1), is not tenable because it doesn't allow us to build module hierarchies where we instantiate `N` copies of the same implementation, chain them together, and prove that the latency is `N*L`. We either need (2) or (3). 

The nice thing about (2) is that we can encode it really efficiently in SMT by saying that each component is a function over parameters:
```
(define-fun (foo Int) Int)
(assert (= F0::L  (foo 10))
(assert (= F1::L  (foo 10))
(assert (= F2::L  (foo 10))
```

This would be sufficient to make the first assertion pass. The downside is that we lose the ability to do instance-specific parameterization but maybe that should be an opt-in feature.

CC @UnsignedByte @gabizon103 @sampsyo

rachitnigam, 2023-10-14:
A slight refinement on the encoding of functions:
```
comp foo[A, B](...) with {
  exists X;
  exists Y;
} { ... }
```
Should get encoded as:
```
(define-fun (foo_X Int Int) Int)
(define-fun (foo_Y Int Int) Int)

(assert (= F0::X (foo_X 10 15))
(assert (= F0::Y (foo_Y 10 15))
...
```

The benefit of this approach is that then we can expose a mechanism to say that particular `exist` parameter is instance-specific which means that compiler should not generate a function to encode their equivalence.

The more I think about this, the better this approach sounds. It has by-default sane semantics and an easy mechanism to allow for opt-in instance-specific parameterization when we decide to support it.

rachitnigam, 2023-10-15:
While we're at this change, let's make some changes to the syntax for existentials:
```
comp Foo[A](...) with {
  some E;      // Defines the function for equality based on inputs
  opaque O;  // Instance-specific
} {
  E := 10;       // Provide binding for E
  O := 20;      // Provide binding for O
}
```

gabizon103, 2023-10-15:
I also like (2) as a solution to this. One thing that might help me understand better, though, is what exactly instance-specific parameterization means?

rachitnigam, 2023-10-15:
This is something we don't support in the language yet but imagine that you have a multiplier defined like this:
```
comp Mult(...) -> (out: ['G, 'G+L] 32) with {
  opaque L
} { ... }

comp main(...) {
  M0 := new Mult;
  M1 := new Mult;
  m0 := M0<'G>(a, b)
  m1 := M1<'G + M0::L>(m0.out, d)
}
```

Imagine, that in this program, you want the instance for `m0` to have a latency of 3 and `m1` to have 4. In this case, even though both of them take the same input parameters (i.e., no parameters), the corresponding existential parameters have different values. 

This is how we reason about *all* existential parameters currently but it seems like a case that you rarely want to use and so should probably be opt-in.

PR #363 (2023-10-15): Opaque and transparent existential parameters
Fixes #354. Implements the proposed syntax changes from the issue as well.
closed 2023-10-15 by rachitnigam