cucapra/filament #382: Merging `hoist-facts` and `discharge` passes

UnsignedByte, 2023-10-20:
Currently, `hoist-facts` takes existing assumptions and moves them to the toplevel by adding path conditions to every assumption, turning it into an implication. This is only useful in `discharge` as it makes it easier to convert these assumptions into SMT. However, this also means that assumptions/assertions become pretty unusable from there onward, as they lose their locality within the code and become hard to monomorphize, etc.

I propose we merge the two passes instead, so that assertions are kept where they are and not changed by `discharge`.

This might also have implications for speed - instead of adding the same path conditions to many assertions, we can instead add path conditions to a new level in the stack when entering a for loop (`0 < i < bound`) for example, and then popping them off the stack. This might speed up discharge, although I'm not sure whether it actually will.

rachitnigam, 2023-10-21:
Historical note: the original compiler did in fact merge discharge and hoist-facts into one. The idea with the new design is:
(1) reduce the duplication in generating path condition prefixes. This is done by interning the `prop`s generated and nothing to do with the separation
(2) Potentially enable simplification of propositions (using the `Simplify` pass) before we send them to the solver

In practice, (2) is disabled because it sometimes removes required `assume` propositions needed to make the solver go fast.

I think one approach to this, which would let us keep the separate pass infrastructure for a future version of more aggressive simplification, is one where we turn hoist facts into a library: something that takes a `&mut ir::Control` and returns all the props that need to be proved.

This would also let us get rid of the destructive updates made by discharge which is useful for post monomorphize checking.

UnsignedByte, 2023-10-21:
What's an `ir::Control`? Is this something new? I can't seem to find it in the code.

What do you mean by duplication of path condition prefixes? Could discharge not generate the path conditions on the fly in the same way as hoist-facts and intern them?

rachitnigam, 2023-10-22:
I meant `ir::Command` (the equivalent is called `ir::Control` in calyx so I mixed them up). And yeah, it can intern the prefixes in the same way that hoist-facts does (that's what I was saying when I said that "interning has nothing to do with the separation of passes).

closed 2023-10-30 by UnsignedByte