cucapra/filament #384: Docs link to friscv is broken

cbiffle, 2023-10-24:
Hi! I was excited to discover Filament, and was looking for a worked example larger than the ALU in the tutorial.

Unfortunately the `friscv` design linked from [the homepage](https://filamenthdl.com/) is a 404, and I can't find it elsewhere in the repo.

Thought I'd let you know in case it broke silently.

rachitnigam, 2023-10-24:
Ah, thanks for letting us know! We moved it under the `apps/` directory which cause the link to break.

cbiffle, 2023-10-24:
Ah, I expected it under `examples/` and didn't check `apps`. What's the difference between an app and an example?

rachitnigam, 2023-10-24:
The `examples/` directory contains stuff that is used in the documentation examples (https://filamenthdl.com/docs/lang/tutorial.html#saving-values-for-the-future) while `apps/` contains bigger designs

rachitnigam, 2024-03-09:
Thanks @Timmmm! We need to revamp the website with the new compiler + syntax anyways so will open a new issue about it.

PR #385 (2023-10-24): webpage link to frisc
Fixes #384.