cucapra/filament #419: Instance borrow syntax should support references to instance's existential parameters

gabizon103, 2024-02-23:
In #413 we discussed wanting to be able to do instance borrows as follows:

`M := new Multipliers[W, N] in ['G, 'G + reuses*M::II + M::L]`

Currently the compiler throws an error at the first use of `M`, saying `M` is not defined. This seems like a pretty useful thing, since the amount of time we want to borrow an instance for kind of necessarily depends on the latency and throughput of the instance.

rachitnigam, 2024-02-23:
Is this a problem in AST to IR conversion?

gabizon103, 2024-02-23:
That's what I was guessing, though I haven't actually checked yet

rachitnigam, 2024-02-23:
Okay, let's triage this. If that is the problem then it should be easy to fix.

gabizon103, 2024-02-23:
Will do and report back 🫡

PR #420 (2024-02-23): Instance borrow ranges can reference instance
Closes #419 
Ended up being a little tweak in `astconv`. When we initially construct the `ir::Instance`, I had it use an empty `Vec` for the `lives` field, and then fill it in after we've already made the source-level name to idx binding in the ctx. This will let us use an instance's existential parameters in its borrow range.

Also adds two small tests, one where it typechecks and one where it doesn't because there aren't any constraints on the existential parameter.
closed 2024-02-23 by gabizon103