cucapra/filament #451: update docs for how to test the Filament installation

CrepeGoat, 2024-09-02:
the "getting started" page says to run `cargo run -- tests/compile/par.fil` to test the build.

doing so I get a few errors:
```sh
$ cargo run -- ./tests/compile/par.fil
error: `cargo run` could not determine which binary to run. Use the `--bin` option to specify a binary, or the `default-run` manifest key.
available binaries: fil-gen, filament
$ cargo run --bin=filament -- ./tests/compile/par.fil
warning: ...
warning: ...

Error: invalid file: Failed to read ./tests/compile/par.fil: No such file or directory (os error 2): 
Compilation failed with 1 errors.
Run with --show-models to generate assignments for failing constraints.
```

I can't find a file named `./tests/compile/par.fil` either. so it seems that the docs should:
1. include the `--bin=filament` flag, and
2. list a working test file

CrepeGoat, 2024-09-02:
as mentioned in https://github.com/cucapra/filament/issues/447#issuecomment-2308932319, I'm asking for this specifically because I'm trying to write [a nix derivation](https://nixos.org) to build and install filament (in my case, specifically on macOS, but this would be applicable to linux builds as well). generally it's good practice to provide tests for a nix build to validate the install, and it's not clear to me what that would look like for Filament.

-> this docs fix (in addition to providing better documentation) would also enable a potential nix build definition to include tests, making the build more resilient to introducing build errors.

rachitnigam, 2024-09-02:
> generally it's good practice to provide tests for a nix build to validate the install, and it's not clear to me what that would look like for Filament.

Filament has a [`runt` test suite](https://github.com/rachitnigam/runt) which is executed using the following steps (after Filament has been configured):
- `cargo install runt` to install the latest `runt` binary
- Typing `runt` in the root of the directory

If it is allowed, I would recommend running the `runt` test suite for the nix package.

CrepeGoat, 2024-09-03:
side note: running `runt` for tests produces test errors. I made a separate issue for this: #452

PR #499 (2025-04-30): Fix getting started instructions
The instruction suggest using a file that no longer exists. I tested this file and it works.

Fixes #451
commit 2d1996e7 (2025-08-08): Fix getting started instructions (#499)

The instruction suggest using a file that no longer exists. I tested this file and it works.

Fixes #451
closed 2025-04-30 by rachitnigam