cucapra/filament #469: Discharge fails if any indices have been invalidated.

UnsignedByte, 2024-10-17:
Discharge works with `DenseIndexInfo`s, meaning if any parameters/events/ports have been invalidated before discharge this throws an error. The culprit is here:
https://github.com/cucapra/filament/blob/2fe2b317fc96ed6453dada529c08289ad08ec4f6/crates/filament/src/ir_passes/discharge.rs#L591-L608

This causes issues because `.ports` and `.events` automatically only iterate over valid indices. We can solve this in two ways:
*  Some sort of "cleanup" pass that removes invalid indices (likely having to reconstruct the entire IR)
* Change `.push` to extend the `DenseIndexInfo` by default. This does not work in the case of discharge though, where `SExpr` is not `Default`.

UnsignedByte, 2024-10-18:
Even larger issue now is that `Expr`s can exist that refer to invalid parameters that have been deleted... Classic dangling pointers.

rachitnigam, 2024-10-19:
Yup, arena allocators are hard... Maybe time to switch to a crate for this or something.

closed 2024-10-22 by UnsignedByte